mavlink/MAVSDK-Python #686: install mavsdk on py zero

naorwaiss, 2024-04-21:
hi ' i try to install mavsdk python on my pi zero - and32 
it not gave me o comp;it the isntallation - stuck at:
  Running setup.py install for grpcio ... /

tanks for the help

JonasVautherin, 2024-04-22:
Feels like an issue with grpcio rather than MAVSDK, then :thinking:

julianoes, 2024-04-23:
Coupe of things wrong here. First of all, I don't recommend running pip with sudo. Running it with sudo means it can interfere with Python packages installed by the package manager.

Second of all, when installing grpcio using pip it doesn't find binaries on PyPi, so it starts building it which will take a while. Therefore, I suggest to use the Debian/Ubuntu packaged versions which are already compiled.

Unfortunately, as I'm discovering, we didn't add this step to the documentation. The steps are in a couple of other issues though:
https://github.com/mavlink/MAVSDK-Python/issues/345#issuecomment-860462360

Essentially, I suggest you do:

```
sudo pip3 uninstall grpcio
pip3 uninstall grpcio
sudo apt-get install python3-grpcio
```

julianoes, 2024-04-23:
Docs update coming: https://github.com/mavlink/MAVSDK-Python/pull/687/commits/48208b9203821e94446bc3e7f162beec8cb90336

julianoes, 2024-05-21:
"Fixed" with docs update.

closed 2024-05-21 by julianoes