mavlink/MAVSDK-Python #691: The use of drone.offboard.set_position_ned

jupiterMJM, 2024-05-08:
Hey everyone,
I want to move the drone one meter to the east (for exemple). As the documentation does not help very much, I thought that I could do that with drone.offboard.set_position_ned . However, when I simulate it under gazebo, it does not work very well and does strange stuff.
Therefore, did anyone yet use it? Or know how to do the thing I wanna do?
Thanks in advance!

julianoes, 2024-05-08:
What do you mean with strange stuff?

One thing to note is that it is relative to your current position. Also see: https://github.com/mavlink/MAVSDK-Python/issues/684

jupiterMJM, 2024-05-09:
Thanks for your reply! After, digging into the log through QGroundControl, I've think I've figured out that drone.offboard.set_position_ned(0, 0, 0, 0) was kinda equivalent to "go back home". Which is not what I wanna do! What I wanna do is:
1/ arm; takeoff the drone and go to a random location (thx to goto_location)
2/ hold for a few second there
3/ move the drone meter after meter according a custom pattern along North, East and Down.
Therefore my questions are:
- is drone.offboard.set_position is done for that? or exists another way to do such thing?
- to initiate the drone.offboard I have to provide initial NED coordinates. To do so, and whithout moving the drone, do I have to do: drone.offboard.set_position_ned(current_north_position_in_NED, current_east_position_in_NED,current_down_position_in_NED). If so, how can I get those info?
- is there a way to make "set the new NED home here"?
Thanks a lot for your answers

jupiterMJM, 2024-05-09:
Okay, so I've succeeded into making such a thing BUT i can't provide the speed I want to apply during the move. Attached, you will find a piece of code (to move) and the picture of the trajectory.
```python
# fonction pour bouger relativement au drone selon le nord/sud/est/ouest
async def move_by(drone:System, north:float, east:float, down:float, yaw:float = 0):
    """
    ATTENTION, FONCTION A EFFET DIRECT SUR LE DRONE; NE PAS UTILISER LA FONCTION SANS REFLECHIR AUX CONSEQUENCES
    """
    global position_du_drone_souhaitee
    print("yo")
    ajout = [north, east, down]
    if await drone.offboard.is_active():
        position_du_drone_souhaitee = [position_du_drone_souhaitee[i] + ajout[i] for i in range(len(position_du_drone_souhaitee))]
        next_point = PositionNedYaw(*position_du_drone_souhaitee, yaw)
        speed_ned = VelocityNedYaw(1, 0, 0, 60)
        accel_ned = AccelerationNed(1, 0, 0)
        await drone.offboard.set_position_velocity_ned(next_point, speed_ned)
        # await drone.offboard.set_position_ned(next_point)
    else:
        print("offboard non activated")
```
[image]
The color gives the speed in m/s.

julianoes, 2024-05-09:
Right, that's always a question: how to set the speed, and unfortunately I don't have a good answer because the normal parameters are not applied in offboard control.

One way is to use velocity control but that requires to close the loop on the outside which is not always what you want to do.

@Jaeyoung-Lim do you know of a way to limit the speed? Could we add a param?

jupiterMJM, 2024-05-10:
Mmmh; okay. But if we can't set the speed what is the aim of the function drone.offboard.set_position_velocity_ned ? I thought that it was a way to provide a target point and the speed to go with to.

julianoes, 2024-05-20:
The speed is a feed forward speed control, not the max speed. I would find max speed more intuitive as well but that's what PX4 exposes, unfortunately.

julianoes, 2024-05-27:
It's the feed foward of the controller. https://en.wikipedia.org/wiki/Feed_forward_(control) I think it's unfortunate because it's not beginner friendly, but it does allow for more aggressive flying.

arpitpara, 2024-06-14:
> Thanks for your reply! After, digging into the log through QGroundControl, I've think I've figured out that drone.offboard.set_position_ned(0, 0, 0, 0) was kinda equivalent to "go back home". Which [...]

Why don't you try offboard from csv example?
I've been working with off-board mode from quite a while now and it seems to work fine for me with GPS and GPS-denied environments to with minimal changes.