mavlink/MAVSDK-Python #692: Unable to upload rally_items through MAVSDK

mfattori, 2024-05-11:
Hi,

I am working with the MissionRaw plugin and am attempting to upload a mission file which also contains rally points to the drone. While I am able to upload a `.plan` file and view the correct mission details through AMC, attempting to perform the same action with MAVSDK-Python ignores the `rally_items`:

## Minimal example:

```python
... # connect to drone, etc 

out = await drone.mission_raw.import_qgroundcontrol_mission("sample_mission.plan")

await drone.mission_raw.upload_rally_points(out.rally_items) # upload rally items -> this is ignored
await drone.mission_raw.upload_mission(out.mission_items) # upload mission items -> this works!
...
```

## Versions:
- MAVSDK Python: mavsdk==2.5.1
- Autopilot: Auterion Skynode:
    - Release Name == v2.15.10
    - AuterionOS == v2.15.10
    - APX4 == 2.7.24
- mavlink-router: v3-11-g42529d5

Am I using the right interface for this and is there an alternate order I should be performing the uploads?

julianoes, 2024-05-20:
Hmm, when you say "ignores" what happens? Does it not upload anything? Do you get an error?

mfattori, 2024-05-20:
No error occurs, it just doesn't appear to upload the rally items to the drone. I can still run the mission items and the drone will go to the waypoints but it won't ever go to the rally points and they aren't visible in AMC.

julianoes, 2024-05-21:
Can you check whether the rally point import actually works. just check the length of the `out.rally_items`. And if it is not working, can you share an example plan that I can use to test the import with?

mfattori, 2024-05-21:
The rally import works, printing the length of `out.rally_items` returns `2`, which is the number of rally items in this mission file.

For reference, the code I am running is:

```python
# ...connect to drone
out = await drone.mission_raw.import_qgroundcontrol_mission("sample_mission.plan")
print(len(out.rally_items)) # prints 2

await drone.mission_raw.upload_rally_points(out.rally_items)
await drone.mission_raw.upload_mission(out.mission_items)
```

This is the mission plan I am testing with:

```json
{
    "UUID": "c79a2c853872309b8dd462791474cb323d936af2",
    "fileType": "Plan",
    "geoFence": {
        "circles": [
        ],
        "polygons": [
            {
                "inclusion": true,
                "polygon": [
                    [
                        43.85722818608631,
                        -79.61147095948077
                    ],
                    [
                        43.85722818608631,
                        -79.60492835311206
                    ],
                    [
                        43.85429874744504,
                        -79.60492835311206
                    ],
                    [
                        43.85429874744504,
                        -79.61147095948077
                    ]
                ],
                "version": 1
            }
        ],
        "version": 2
    },
    "groundStation": "QGroundControl",
    "mission": {
        "cruiseSpeed": 15,
        "firmwareType": 12,
        "globalPlanAltitudeMode": 1,
        "hoverSpeed": 5,
        "items": [
            {
                "MISSION_ITEM_ID": "0",
                "autoContinue": true,
                "command": 178,
                "doJumpId": 1,
                "frame": 2,
                "params": [
                    1,
                    5,
                    -1,
                    0,
                    0,
                    0,
                    0
                ],
                "type": "SimpleItem"
            },
            {
                "AMSLAltAboveTerrain": 100,
                "Altitude": 100,
                "AltitudeMode": 1,
                "MISSION_ITEM_ID": "2",
                "autoContinue": true,
                "command": 16,
                "doJumpId": 2,
                "frame": 3,
                "params": [
                    0,
                    0,
                    0,
                    null,
                    43.85613586,
                    -79.60898841,
                    100
                ],
                "type": "SimpleItem"
            },
            {
                "AMSLAltAboveTerrain": 100,
                "Altitude": 100,
                "AltitudeMode": 1,
                "MISSION_ITEM_ID": "3",
                "autoContinue": true,
                "command": 16,
                "doJumpId": 3,
                "frame": 3,
                "params": [
                    0,
                    0,
                    0,
                    null,
                    43.85572471,
                    -79.60793674,
                    100
                ],
                "type": "SimpleItem"
            },
            {
                "AMSLAltAboveTerrain": 100,
                "Altitude": 100,
                "AltitudeMode": 1,
                "MISSION_ITEM_ID": "4",
                "autoContinue": true,
                "command": 16,
                "doJumpId": 4,
                "frame": 3,
                "params": [
                    0,
                    0,
                    0,
                    null,
                    43.85642823,
                    -79.60695476,
                    100
                ],
                "type": "SimpleItem"
            },
            {
                "AMSLAltAboveTerrain": 100,
                "Altitude": 100,
                "AltitudeMode": 1,
                "MISSION_ITEM_ID": "5",
                "autoContinue": true,
                "command": 16,
                "doJumpId": 5,
                "frame": 3,
                "params": [
                    0,
                    0,
                    0,
                    null,
                    43.85687592,
                    -79.6084499,
                    100
                ],
                "type": "SimpleItem"
            },
            {
                "autoContinue": true,
                "command": 20,
                "doJumpId": 6,
                "frame": 2,
                "params": [
                    0,
                    0,
                    0,
                    0,
                    0,
                    0,
                    0
                ],
                "type": "SimpleItem"
            }
        ],
        "plannedHomePosition": [
            43.856405656176875,
            -79.60898841,
            217
        ],
        "vehicleType": 2,
        "version": 2
    },
    "rallyPoints": {
        "points": [
            [
                43.85535468,
                -79.6069801,
                0
            ],
            [
                43.8566064,
                -79.60628322,
                0
            ]
        ],
        "version": 2
    },
    "version": 1
}
```

Thank you!

julianoes, 2024-05-21:
Great, thanks for that.

julianoes, 2024-05-23:
Ok, I tried this and I get this:

```
  File "/home/julianoes/src/upstream/MAVSDK-Python/examples/mission_import.py", line 32, in <module>
    asyncio.run(run())
  File "/usr/lib/python3.10/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.10/asyncio/base_events.py", line 649, in run_until_complete
    return future.result()
  File "/home/julianoes/src/upstream/MAVSDK-Python/examples/mission_import.py", line 25, in run
    await drone.mission_raw.upload_rally_points(mission_import_data.rally_items)
  File "/home/julianoes/.local/lib/python3.10/site-packages/mavsdk/mission_raw.py", line 831, in upload_rally_points
    raise MissionRawError(result, "upload_rally_points()", mission_items)
mavsdk.mission_raw.MissionRawError: BUSY: 'Busy'; origin: upload_rally_points(); params: ([<mavsdk.mission_raw.MissionItem object at 0x7f77f831b400>, <mavsdk.mission_raw.MissionItem object at 0x7f77f831b430>],)
```

As soon as I add in a sleep before doing the rally points, it starts working. That's a workaround but hopefully gets you unblocked.

```
    await drone.mission_raw.upload_mission(out.mission_items)
    await asyncio.sleep(0.1)
    await drone.mission_raw.upload_rally_points(out.rally_items)
```

julianoes, 2024-05-23:
And the real fix is over here: https://github.com/mavlink/MAVSDK/pull/2311

closed 2024-05-23 by julianoes

julianoes, 2024-05-26:
They were uploaded in my testing just fine, so that's confusing.

julianoes, 2024-05-26:
One other thing to consider is whether your version of PX4 has this: https://github.com/PX4/PX4-Autopilot/pull/22309
If you're using v1.14.0 it will not. If you use v1.14.3 it would include it.