mavlink/MAVSDK-Python #695: Error when uploading mission: INVALID_PARAM1

ShafSpecs, 2024-05-13:
Whenever I try to upload a mission via `vehicle.mission.upload_mission()`, even using the examples provided in this repository, I get the following mavlink error:
```
Got COMMAND_ACK: REQUEST_MESSAGE: FAILED
Got MISSION_ACK: TYPE_MISSION: INVALID_PARAM1
```
I am using a simple:
```py
mission_list = []

mission_list.append(
                MissionItem(47.398039859999997,
                            8.5455725400000002,
                            25,
                            10,
                            True,
                            float('nan'),
                            float('nan'),
                            MissionItem.CameraAction.NONE,
                            float('nan'),
                            float('nan'),
                            float('nan'),
                            float('nan'),
                            float('nan'),
                            MissionItem.VehicleAction.NONE))

mission_plan = MissionPlan(mission_list)
vehicle.mission.upload_mission(mission_plan)
```

JonasVautherin, 2024-05-13:
Are you using PX4 or Ardupilot?

ShafSpecs, 2024-05-14:
> Are you using PX4 or Ardupilot?

ArduPilot

ShafSpecs, 2024-05-14:
Now that you mention it, is this an issue with how ArduPilot and PX4 handle mission sequences? Meaning, do I need to send it in raw format?

JonasVautherin, 2024-05-14:
I think so, yes :+1:

ShafSpecs, 2024-05-14:
> I think so, yes :+1:

Thank you. Question though, I know one of them has a sequence starting with 1 and the other one's starting sequence is zero. Do you know which one PX4/ArduPilot is?

ShafSpecs, 2024-05-14:
Seems to be zero. Thank you very much.