mavlink/MAVSDK-Python #703: Running missions

ShafSpecs, 2024-05-25:
Hello, question about nature of missions in MAVSDK (Ardupilot). I am using the raw mission to create and upload missions. I have a set of waypoints and a takeoff command ([command 22](https://mavlink.io/en/messages/common.html#MAV_CMD_NAV_TAKEOFF)). I upload mission fine, no problem. But when I want to start a mission (arm, then start mission):
```py
await vehicle.action.arm()
await vehicle.mission_raw.start_mission()
```
I get an error in my SITL console:
```
AP: Auto: Missing Takeoff Cmd
AP: Mode change to AUTO failed: init failed
```
Even though my very first mission item is:
```py
RawMissionItem(
                seq=0,
                frame=3,
                command=22,
                current=1,
                autocontinue=1,
                param1=10,
                param2=0,
                param3=0,
                param4=float('nan'),
                x=0,
                y=0,
                z=float(10.0),
                mission_type=0
)
```
I am not sure what I am missing here. Any help please?

JonasVautherin, 2024-05-25:
> AP: Auto: Missing Takeoff Cmd

Doesn't it mean that you need to add a takeoff item in the mission?

ShafSpecs, 2024-05-25:
> > AP: Auto: Missing Takeoff Cmd
> 
> Doesn't it mean that you need to add a takeoff item in the mission?

Yeah. That's the first mission item I added. MAV_CMD_NAV_TAKEOFF translates to command 22 but for some reason, it still says it's missing

JonasVautherin, 2024-05-25:
You should check what Ardupilot receives, I think. Because that message is coming from Ardupilot, right? Apparently Ardupilot is not happy with the MAVLink mission item you constructed (presumably it does not recognize the takeoff item).

ShafSpecs, 2024-05-25:
Okay. Would debug and get back to you. Thanks

ShafSpecs, 2024-05-26:
> You should check what Ardupilot receives, I think. Because that message is coming from Ardupilot, right? Apparently Ardupilot is not happy with the MAVLink mission item you constructed (presumably it does not recognize the takeoff item).

Did some digging, and I think the problem is from Ardupilot. Or a protocol mismatch between ardupilot and mavsdk. Mission sequence starts at 1 within ardupilot: https://discuss.ardupilot.org/t/planning-missions-via-mavlink/16489/8 ([Full thread](https://discuss.ardupilot.org/t/planning-missions-via-mavlink/16489/17)). My mission planning always started from 0. 

To debug, I appended the takeoff command twice. With sequence 0 and sequence 1, then the RTL command as number 2. It worked. So apparently, the sequence 0 is meant to be home command? Still looking into that but thanks. I might just extend mavsdk python on my end and create an AP plugin. Like a wrapper around the core, existing ones. Thanks again ❤️

ShafSpecs, 2024-05-26:
Closing this as solved. Would still add any extra info tidbit I encounter to this thread to assist others

closed 2024-05-26 by ShafSpecs

julianoes, 2024-05-27:
That's correct. Home, or just a 0 entry is required for ArduPilot.

ShafSpecs, 2024-05-27:
> That's correct. Home, or just a 0 entry is required for ArduPilot.

What is a 0 entry? I think that sounds like a better option. Because the vehicle sets its home to its location everytime an arm occurs. Might cause issues if I upload a mission in point A (using the location of A) and the drone is moved to B before mission execution

ShafSpecs, 2024-05-27:
And when you say home, do you mean HOME_POSITION ([242](https://hamishwillee.gitbooks.io/ham_mavdevguide/content/en/messages/common.html#HOME_POSITION)) or a set home position command?

julianoes, 2024-05-27:
I'm not sure to be honest. I think safest is to just put a NAV item with everything 0 in there. Or just try out in SITL whether you can figure out what happens.

ShafSpecs, 2024-05-27:
> I'm not sure to be honest. I think safest is to just put a NAV item with everything 0 in there. Or just try out in SITL whether you can figure out what happens.

Would do just that. Probably set two takeoff commands but set the one at sequence 1 to current. So it starts from there automatically 👍. Thanks a lot

ShafSpecs, 2024-05-28:
Looking at Ardupilot mission planner, seems they just did that 😆 
```
QGC WPL 110
0	1	0	16	0	                0	                0	                0	                6.4253317	3.4094982	0.070000	      1
1	0	3	22	0.00000000	0.00000000	0.00000000	0.00000000	0.00000000	0.00000000	10.000000      1
2	0	3	16	10.00000000	0.00000000	0.00000000	0.00000000	6.42555240	3.40962140	30.000000      1
3	0	3	20	0.00000000	0.00000000	0.00000000	0.00000000	0.00000000	0.00000000	0.000000	       1
```
Although, it's waypoint at sequence 0, not takeoff. And I am more concerned about why their MISSION_TYPE parameter is set to geofence for each item instead of a mission item 🤔, or maybe I am reading that last parameter wrong. Looking at it again, it might be the autocontinue param