mavlink/MAVSDK-Python #704: How to use mavsdk-python to takeoff without GPS

Hmilzy, 2024-05-31:
Hello ! 
I will first introduce the framework. I have a control PC and a drone connected to the Raspberry Pi via a serial port.
Now I want to send a "takeoff" command from my PC and transmit it to the Raspberry Pi via UDP. After receiving the command, the Raspberry Pi will execute the program drone.action.takeoff() in mavsdk-python to control the takeoff.
I deployed mavsdk-server and mavlink-router on the Raspberry Pi.
mavlink-routerd -e  127.0.0.1:30000 /dev/ttyAMA0:921600          ./mavsdk_server ....  -p 50040 udp://0.0.0.0:30000

I conducted an experiment indoor，I was able to successfully connect to the drone and commands were sent from the control computer to the Raspberry Pi. But when drone.action.takeoff() is executed, an error like command denied will be reported. 

So I want to know whether the prerequisite for takeoff is to receive GPS signal, because I can successfully arm indoors, but I can't takeoff.

And If I want to control the flight indoors, how should I write mavsdk-python?

julianoes, 2024-05-31:
That's a question that comes up often. I think it boils down to whether PX4 (assuming you're using PX4) can do autonomous flight without GPS.

Usually you need something like flow or VIO or a MOCAP system to do that. Otherwise you have to fly manually because you'd drift away pretty quickly.

Hmilzy, 2024-05-31:
Thank you for your answer.
Is it possible to use offboard mode?

`
    await drone.action.arm()

    print("Setting initial setpoints...")
    await drone.offboard.set_velocity_ned(
        VelocityNedYaw(0.0, 0.0, 0.0, 0.0))  

    print("Starting offboard control...")
    try:
        await drone.offboard.start()
    except OffboardError as error:
        print(f"Starting offboard mode failed with error code: {error._result.result}")
        print("-- Disarming")
        await drone.action.disarm()
        return

    print("Taking off manually...")
    await drone.offboard.set_velocity_ned(
        VelocityNedYaw(0.0, 0.0, -1.0, 0.0))  # Ascend with -1.0 m/s velocity

    await asyncio.sleep(5)

    print("Hovering...")
    await drone.offboard.set_velocity_ned(
        VelocityNedYaw(0.0, 0.0, 0.0, 0.0))  # Hover in place

    await asyncio.sleep(10)

    print("Landing manually...")
    await drone.offboard.set_velocity_ned(
        VelocityNedYaw(0.0, 0.0, 1.0, 0.0))  # Descend with 1.0 m/s velocity

    await asyncio.sleep(5)

    print("Stopping offboard control...")
    try:
        await drone.offboard.stop()
    except OffboardError as error:
        print(f"Stopping offboard mode failed with error code: {error._result.result}")

    print("Disarming...")
    await drone.action.disarm()
`

julianoes, 2024-05-31:
No, offboard requires a position (or at least velocity) estimate, unless you use attitude offboard.

arpitpara, 2024-06-14:
It's possible through off-board mode. As @julianoes mentioned without GPS drones should have either VIO or MO-CAP.
We've built a indoor drone using VIO with PX4. We use Velocity-Ned and Position-NED(Prefered) function for autonomous control through MAVSDK-Python. You can also try it.