mavlink/MAVSDK-Python #708: Analyzing log files or Distance travelled data from mavsdk python

arpitpara, 2024-06-12:
Hi
I've been wondering if we can calculate distance traveled by the vehicle in real-time and plot it live.
Also, if we can analyze the logs of flight through the same.

julianoes, 2024-06-14:
You will have to subsribe to position updates and do the integration yourself.

arpitpara, 2024-06-14:
I am working on GPS-Denied environments. So I can't calculate it with GPS. I working with VIO .

julianoes, 2024-06-14:
Use the local position updates: http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/plugins/telemetry.html#mavsdk.telemetry.PositionVelocityNed

arpitpara, 2024-06-15:
How is odometry is different from PositionVeocityNed

julianoes, 2024-06-19:
Odometry would be the info from an odometry system. Position velocity come from the autopilot's estimator.

closed 2024-06-19 by julianoes