mavlink/MAVSDK-Python #711: connectivity

peakyquest, 2024-06-20:
```
#!/usr/bin/env python3

import asyncio
from mavsdk import System

async def run():
    # Connect to the drone
    drone = System()
    await drone.connect('serial://com7:57600')

# Run the asyncio loop
asyncio.run(run())

```

what I am doing wrong ?

peakyquest, 2024-06-21:
I am working on windows

ShafSpecs, 2024-06-21:
> I am working on windows

On Windows, you would need to actually start mavsdk_server manually and then pass the server port to the System instantiation. I'm not near my PC atm so I can't paste a snippet

julianoes, 2024-06-22:
Try the debugging instructions: http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/#debug-connection-issues

ShafSpecs, 2024-06-22:
Or check out this comment: https://github.com/mavlink/MAVSDK-Python/issues/678#issuecomment-2065319944