mavlink/MAVSDK-Python #713: Support custom COMMAND_LONG listeners and be able to ACK

sanderux, 2024-06-24:
We are using MAV_CMD_WAYPOINT_USER_1 to run custom commands from a mission
This ability is currently missing from mavsdk making it impossible for us to switch.
Would be nice to be able to add a custom action listener.

@julianoes FYI

julianoes, 2024-06-24:
I think this should be possible using the passthrough plugin. Before I add it as explicit functionality we would have to standardize it in MAVLink. We had several discussions about custom action commands before, the previous attempt was a bit too complex though, in my opinion: https://github.com/mavlink/MAVSDK/pull/1581#issuecomment-1092265844.

sanderux, 2024-06-24:
User actions are standardized in mavlink no?
What would be wrong with extending the actions call with a 'custom action' handler where you can listen to all commands and choose if you want to handle / ack them?

sanderux, 2024-06-24:
Here: mavsdk/action_server.py

julianoes, 2024-06-25:
Does the autopilot (e.g. PX4) support sending USER_1 commands and waiting for an ack? That would be the first step.

sanderux, 2024-06-27:
Yes it does, we are already using this with pymavlink

julianoes, 2024-06-28:
Ok, which PX4 version do I use to test this?