mavlink/MAVSDK-Python #714: can't connect to QGC via UDP connection using mavsdk

hilaelb, 2024-06-27:
I want to connect to QGC, i using this code, but QGC still shows on the main screen disconnected. why is this happen?

import asyncio
from mavsdk import System

async def run():
    # Create a MAVSDK system object
    drone = System()

    # Connect to the drone via QGC's UDP port (assuming QGC is configured to listen on 14550)
    await drone.connect(system_address="udp://192.168.168.95:14550")

    print("Waiting for drone to connect...")
    async for state in drone.core.connection_state():
        if state.is_connected:
            print(f"Drone discovered with UUID: {state.uuid}")
            break

    print("Drone connected!")

    # Listen for heartbeat messages
    async for heartbeat in drone.telemetry.heartbeat():
        print(f"Heartbeat: {heartbeat}")

if __name__ == "__main__":
    loop = asyncio.get_event_loop()
    loop.run_until_complete(run())

[image]
[image]

JonasVautherin, 2024-06-27:
Can you elaborate on your use-case? Do you want MAVSDK to behave like an autopilot? Because your code above tries to connect MAVSDK as a ground station to QGC, which won't work.

hilaelb, 2024-06-29:
> Can you elaborate on your use-case? Do you want MAVSDK to behave like an autopilot? Because your code above tries to connect MAVSDK as a ground station to QGC, which probably won't work.

Yes, I want that MAVSDK behave like a generic autopilot and I want to connect the QGC, in another words - what do I require for a connection to be established with a generic autopilot?

julianoes, 2024-06-29:
You need to create the autopilot as a plugin. I don't think this has been attempted using a language wrapper (like Python) yet. I'm not certain it will work. I would suggest to do it using MAVSDK in C++ and start with this example:
https://github.com/mavlink/MAVSDK/tree/main/examples/autopilot_server

julianoes, 2025-09-16:
You can now use MavlinkDirect to susbscribe to any message, see this example: https://github.com/mavlink/MAVSDK-Python/blob/main/examples/mavlink_direct_receive.py

closed 2025-09-16 by julianoes