mavlink/MAVSDK-Python #715: Get HEARTBEAT message from MAVSDK

ShafSpecs, 2024-06-28:
I am not sure if this is shipped with MAVSDK, as I didn't see any related docs online. How do we retrieve heartbeat messages ([0](https://mavlink.io/en/messages/minimal.html#HEARTBEAT))? And is there anyway to listen to custom (yet valid and sent) MAVLink messages

> Side question for maintainers: If heartbeat isn't supported, is it a thing of creating a `.proto` file for it? Or it is more complicated?

Thanks 😊!

julianoes, 2024-06-29:
Also, what information of the heartbeat message do you actually need?

ShafSpecs, 2024-06-29:
> Also, what information of the heartbeat message do you actually need?

System status. That's the most critical, I would like to collect timestamps as well using the inboard computer. But the status is the most critical

ShafSpecs, 2024-06-29:
Would dive into the technicalities of the issue when I get back. But at a glance, I prefer the interface method. When sending raw mavlink messages, the checks for validity are done by the vehicle on the fly, not at compile time.

And using interface means whatever new mavlink feature comes up doesn’t require a full minor release or smthg

ShafSpecs, 2024-06-29:
What is Mavlink passthrough please? Sending raw mavlink messages, I assume.
I had started writing hypothetical APIs and how it would work, when I didn't know how the passthrough plugin would look like in practice 😅.

Btw, do you know [pymavlink](https://github.com/ArduPilot/pymavlink/)? A great example of keeping it low-level with higher-level utilities and safeguards

closed 2025-09-16 by julianoes