mavlink/MAVSDK-Python #717: MAVSDK server not connecting with JMAVSIM - Windows WSL

nehal-rehman, 2024-07-09:
I am running jmavsim on wsl and the GUI seems to work fine with the wsl commands to takeoff and land but I am trying to connect it to the MAVSDK server and its not connecting

here are my logs and code:

px4 sitl mode for jmavsim

[1/2] launching px4 jmavsim_iris (SYS_AUTOSTART=10017)

______  __   __    ___
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] startup script: /bin/sh etc/init.d-posix/rcS 0
env SYS_AUTOSTART: 10017
INFO  [param] selected parameter default file parameters.bson
INFO  [param] importing from 'parameters.bson'
INFO  [parameters] BSON document size 316 bytes, decoded 316 bytes (INT32:13, FLOAT:3)
INFO  [param] selected parameter backup file parameters_backup.bson
INFO  [dataman] data manager file './dataman' size is 7872608 bytes
INFO  [init] jMAVSim simulator
INFO  [simulator_mavlink] using TCP on remote host localhost port 4560
WARN  [simulator_mavlink] Please ensure port 4560 is not blocked by a firewall.
INFO  [simulator_mavlink] Resolved host 'localhost' to address: 127.0.0.1
INFO  [simulator_mavlink] Waiting for simulator to accept connection on TCP port 4560
Buildfile: /mnt/e/PROJECTS_UTF/PX4-Autopilot/Tools/simulation/jmavsim/jMAVSim/build.xml

make_dirs:

compile:

create_run_jar:

copy_res:

BUILD SUCCESSFUL
Total time: 10 seconds
Options parsed, starting Sim.
Starting GUI...
INFO  [simulator_mavlink] Simulator connected on TCP port 4560.
Init MAVLink
INFO  [lockstep_scheduler] setting initial absolute time to 1720553661012000 us
INFO  [commander] LED: open /dev/led0 failed (22)
WARN  [health_and_arming_checks] Preflight Fail: ekf2 missing data
INFO  [uxrce_dds_client] init UDP agent IP:127.0.0.1, port:8888
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 18570 remote port 14550
WARN  [health_and_arming_checks] Preflight: GPS fix too low
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [mavlink] mode: Onboard, data rate: 4000 B/s on udp port 14280 remote port 14030
INFO  [mavlink] mode: Gimbal, data rate: 400000 B/s on udp port 13030 remote port 13280
INFO  [logger] logger started (mode=all)
INFO  [logger] Start file log (type: full)
INFO  [logger] [logger] ./log/2024-07-09/19_34_24.ulg
INFO  [logger] Opened full log file: ./log/2024-07-09/19_34_24.ulg
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [px4] Startup script returned successfully
pxh> WARN  [health_and_arming_checks] Preflight: GPS fix too low
INFO  [tone_alarm] home set
WARN  [health_and_arming_checks] Preflight: GPS fix too low
INFO  [commander] Ready for takeoff!

server logs:

mavsdk_server_win32.exe -p 50051 udp://127.0.0.1:18570
[12:29:12|Info ] MAVSDK version: v0.35.0 (D:\a\MAVSDK\MAVSDK\src\core\mavsdk_impl.cpp:20)
[12:29:12|Debug] New: System ID: 0 Comp ID: 0 (D:\a\MAVSDK\MAVSDK\src\core\mavsdk_impl.cpp:417)
[12:29:12|Info ] Server started (D:\a\MAVSDK\MAVSDK\src\backend\src\grpc_server.cpp:45)
[12:29:12|Info ] Server set to listen on 0.0.0.0:50051 (D:\a\MAVSDK\MAVSDK\src\backend\src\grpc_server.cpp:46)
[12:29:12|Info ] Waiting to discover system on udp://127.0.0.1:18570... (D:\a\MAVSDK\MAVSDK\src\backend\src\connection_initiator.h:21)

I tried running it on every port including the default 14540 and every combination with or without the IP 127.0.0.1

here is my python edited code on takeoff and land example from mavsdk-python

    print("Starting the drone connection script...")

    # Connect to the MAVSDK server
    drone = System(mavsdk_server_address='localhost', port=50051)
    await drone.connect(system_address="udp://:18570")
    
    
Also ran this code on every port but its stuck at connection with drone, earlier it was not even running with x64 version I found someone running it on x32 so shifted to that seems like my python is connecting with it but not the jmavsim
    
logs of takeoff_and_land.py:
Starting the drone connection script...
Waiting for drone to connect...

what could be the problem?

JonasVautherin, 2024-07-10:
Shouldn't you use `udp://:14550` (or `udp://:14540`) instead of `udp://:18750`? 🤔

nehal-rehman, 2024-07-18:
I have used all the ports including udp://:14550 and udp://:14540, this is solved by running the python script within my wsl env. I didn't need to use the webserver separately either.

closed 2024-07-18 by nehal-rehman