mavlink/MAVSDK-Python #723: Manual control of RGB LED during flight using MAVSDK?

alireza787b, 2024-08-19:
I’m working on a project where I need to control the onboard RGB LED (e.g., the one in the GPS module) during certain phases of flight, such as during a drone show. Is there any existing feature in MAVSDK that allows this? If not, do you have any recommendations or insights on how to achieve this, either through MAVSDK or using another method with PX4? I want to avoid external LEDs if possible for now or later on add external LED that obey the onboard Pixhawk LEDs

julianoes, 2024-08-20:
What's the MAVLink message this would use? Presumably we need to add the functionality to MAVSDK.

hamishwillee, 2024-08-21:
FYI there is no MAVLink message to do this, and likely this would not be a good idea for safety reasons. The LEDs have specific meanings - in PX4 docs here https://docs.px4.io/main/en/getting_started/led_meanings.html but same/similar in ArduPilot.

So you'd need a flight stack to agree to disable its meanings, since its drivers are going to be talking to the GPS/LEDs.

You could add support for controlling other LEDs using https://mavlink.io/en/services/illuminator.html

alireza787b, 2024-08-25:
> FYI there is no MAVLink message to do this, and likely this would not be a good idea for safety reasons. The LEDs have specific meanings - in PX4 docs here [...]

so If I understand correctly, right now we have no way of directly changing the lights using mavlink commands. the only method is changing directly in px4 source codes and modules which is not the best practice.  is the illumination for any optional light that can be added as an extra?
in this case, I prefer to have my custom LED strips, control them with a companion computer or PWM outputs actuator commands, and during the drone show disable onboard LEDs so lights won't be conflicting.

julianoes, 2024-08-26:
@hamishwillee actually, illuminator is more for lights to light up a scene or similar, right? Not so much for RGB on the drone.

hamishwillee, 2024-08-26:
@alireza787b Yes. 

> in this case, I prefer to have my custom LED strips, control them with a companion computer or PWM outputs actuator commands, and during the drone show disable onboard LEDs so lights won't be conflicting.

Yes, so you you can control your LED strips through custom methods, or via the illuminator command. But you'll have to implement it in the flight stack or companion. All I'm saying here is that this would be the approach for controlling a custom "lighting source".

Turning off all RGB from the standard LEDs would be IMO unsafe, but you might be able to convince a flight stack to turn off the the "All OK colours" - which is what you'd have when flying your light show. So if you had some problem the lights would still go on.  Or for anything you can create a custom build and do what you like from a "coding" point of view.

Though again, none of the flight stacks are likely to include code to disable those.

> @hamishwillee actually, illuminator is more for lights to light up a scene or similar, right? Not so much for RGB on the drone.

Fair-ish. First  design was for controlling LEDs, then it became more like a torch. But with that new control message it is half way between - i.e. it supports strobing but not complex colour control, fading and all that stuff. https://mavlink.io/en/messages/common.html#MAV_CMD_DO_ILLUMINATOR_CONFIGURE .
So it might not be fit for this kind of purpose ... or it might.

Either way, sounds like @alireza787b is looking at custom builds.