mavlink/MAVSDK-Python #724: Mavsdk in Airflow run issue

gsrinivaskishan, 2024-08-21:
When running Mavsdk as python progam using Airflow, below error is shown 

RuntimeError: {plugin_name} plugin has not been initialized! Did you run `System.connect()`?

Kindly help on fixing the issue

JonasVautherin, 2024-08-21:
I don't know Airflow. Do you run some code? If yes, can you show us the code you run?

gsrinivaskishan, 2024-08-21:
Kindly find the code used this is executed after 

await drone.connect(system_address="udp://:14540")

This line is executed still get above error.

async def print_battery(self,drone):
            for position in drone.telemetry.battery():
                data["battery"] = position.remaining_percent

julianoes, 2024-08-22:
Debug the connection first, see: http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/index.html#debug-connection-issues

closed 2024-08-23 by gsrinivaskishan