mavlink/MAVSDK-Python #728: Command Denied when trying to set imu rate

Tuxliri, 2024-09-04:
I am trying to set the imu rate using `System.telemetry.set_rate_imu()`,

julianoes, 2024-09-04:
Denied would mean that ArduPilot doesn't accept what you're asking. Have you tried a lower rate?

This sounds like an ArduPilot issue, not a MAVSDK one to me.

Tuxliri, 2024-09-12:
Yes I had tried a rate of 1Hz as well with same results.

julianoes, 2024-09-14:
Can you debug the ArduPilot side and check in code what's happening there?

Tuxliri, 2024-09-16:
@julianoes for sure, I don't know what tools are used to debug Ardupilot, do you have some resources I could read about this? Is it a logfile-based debugging or can I run the software on the flight controller and attach a C++ debugger?

julianoes, 2024-09-17:
My first approach would be to try it in SITL. And then my assumption would be that you can use gdb or printfs when you run it in in SITL, at least that's what it would be for PX4.

Tuxliri, 2024-09-17:
Alright, I'll give that a go.