mavlink/MAVSDK-Python #730: PID Controller

DavidNietoIMAGO, 2024-09-26:
Is it possible to modify the controller using Python and MAVSDK? Specifically, I assume the system uses a PID controller by default, but I would like to experiment with other controllers, such as backstepping or nonlinear controllers. Could you provide guidance on how to implement this?

JonasVautherin, 2024-09-26:
Do you mean that you would like to change the implementation of the autopilot (be it Ardupilot or PX4)? In that case, you should probably look into the source code of the autopilot itself. From MAVSDK you will be able to remotely set parameters, but MAVSDK is not the autopilot...

DavidNietoIMAGO, 2024-09-26:
I understand that MAVSDK is not the autopilot. What I want is to modify the controller embedded in the autopilot to test different controllers, adjust their gains, and possibly incorporate disturbance observers—essentially focusing on control engineering aspects. I plan to start with Software-in-the-Loop (SIL) testing and potentially move to testing in an actual aircraft later on.

julianoes, 2024-09-30:
You can use MAVSDK to look at realtime data of your controller, however, you can only observe the outputs but not the controller internal states or just more internal autopilot data since that is not shared via MAVLink but only via uORB. To see the internal data, you're better off using the ROS 2 bridge (https://docs.px4.io/main/en/middleware/uxrce_dds.html) which bridges to uORB.

You can use MAVSDK to change parameters, so to adjust gains.

closed 2024-09-30 by julianoes

DavidNietoIMAGO, 2024-10-02:
Thank you for your quick and accurate answers