mavlink/MAVSDK-Python #731: `MissionRawError`  unknown when starting a mission

Ziaad03, 2024-10-12:
I am encountering a MissionRawError unknown when attempting to start a mission using MAVSDK.  Additionally, I don't know if it is gonna help solve this problem, but I see a related message in the drone console that states: "AP: Mode change to AUTO failed: init failed."
 
 here is the script 
 
 ```python
 
 #!/usr/bin/env python3

import asyncio
from mavsdk import System
from mavsdk import mission_raw
from mavsdk.action_server import FlightMode

async def px4_connect_drone():
    drone = System()
    await drone.connect(system_address="udp://:14540")

    print("Waiting for drone to connect...")
    async for state in drone.core.connection_state():
        if state.is_connected:
            print("-- Connected to drone!")
            return drone

async def run():
    drone = await px4_connect_drone()
    await run_drone(drone)

async def run_drone(drone):

    #clear previous missions
    print("-- Clearing previous missions")
    await drone.mission_raw.clear_mission()

    print("-- Downloading Previous mission")
    downloaded_items = await drone.mission_raw.download_mission()
    print("-- Done")

    mission_items = []

    
   
    # Fly to the first waypoint
    mission_items.append(mission_raw.MissionItem(
        0,  # Sequence number
        3,  # MAV_FRAME_GLOBAL_RELATIVE_ALT
        16,  # MAV_CMD_NAV_WAYPOINT
        1,  # Set current command
        1,  # Auto-continue
        0.0,  # Minimum pitch
        0.0,  # Empty param
        0.0,  # Empty param
        0,  # Yaw angle
        int(47.398039859999997 * 1e7),  # Latitude
        int(8.5455725400000002 * 1e7),  # Longitude
        25.0,  # Altitude
        0  # Mission type
    ))

    
    

    #upload mission
    print("-- Uploading mission")
    await drone.mission_raw.upload_mission(mission_items)
    print("-- Done")

    # Download Mission
    print("-- Downloading mission")
    downloaded_items = await drone.mission_raw.download_mission()
    print("-- Done")
    print(f"Mission size: {len(downloaded_items)}")
    print(downloaded_items)
    
    #Arm drone
    print("-- Arming")
    await drone.action.arm()

    

    #Takeoff
    print("-- Taking off")
    await drone.action.takeoff()

    await asyncio.sleep(10)

    
    #start mission
    print("-- Starting mission")
    await drone.mission_raw.start_mission()

    

if __name__ == "__main__":
    # Run the asyncio loop
    asyncio.run(run())
 
 ```
 
 I don't know if this issue I am having because Mavsdk is not 100% compatible with ardupilot.

julianoes, 2024-10-13:
This has likely to do with your sequence number and the fact that you have to have a 0 issue for the home position.

Ziaad03, 2024-10-13:
I added a takeoff command at the beginning of the mission, and it started and ended successfully. However, I noticed that the drone ignored the takeoff command and instead executed the next mission item, which is after the takeoff command. So I need to use `action.takeoff()` to take off so the drone can complete the mission. 

This is the code I ran

```python

#!/usr/bin/env python3

import asyncio
from mavsdk import System
from mavsdk import mission_raw

async def px4_connect_drone():
    drone = System()
    await drone.connect(system_address="udp://:14540")

    print("Waiting for drone to connect...")
    async for state in drone.core.connection_state():
        if state.is_connected:
            print("-- Connected to drone!")
            return drone

async def run():
    drone = await px4_connect_drone()
    await run_drone(drone)

async def run_drone(drone):

    # Clear previous missions
    print("-- Clearing previous missions")
    await drone.mission_raw.clear_mission()

    

    mission_items = []

    # Get current position (home position) from the GPS
    async for gps_info in drone.telemetry.position():
        latitude = gps_info.latitude_deg
        longitude = gps_info.longitude_deg
        break  # Break after getting the first GPS fix

    print(f"Home position: {latitude}, {longitude}")

    # Add a takeoff command at sequence number 0 (for the home position)
    mission_items.append(mission_raw.MissionItem(
        0,  # Sequence number
        3,  # MAV_FRAME_GLOBAL_RELATIVE_ALT
        22,  # MAV_CMD_NAV_TAKEOFF
        1,  # Set current command
        1,  # Auto-continue
        0.0,  # Minimum pitch
        0.0,  # Empty param
        0.0,  # Empty param
        0,  # Yaw angle
        int(latitude * 1e7),  # Latitude (current position)
        int(longitude * 1e7),  # Longitude (current position)
        25.0,  # Altitude
        0  # Mission type
    ))

    # Fly to the first waypoint (sequence number 1 now)
    mission_items.append(mission_raw.MissionItem(
        1,  # Sequence number
        3,  # MAV_FRAME_GLOBAL_RELATIVE_ALT
        16,  # MAV_CMD_NAV_WAYPOINT
        0,  # Set current command
        1,  # Auto-continue
        0.0,  # Minimum pitch
        0.0,  # Empty param
        0.0,  # Empty param
        0,  # Yaw angle
        int(-35.3593237 * 1e7),  # Latitude
        int(149.1679645 * 1e7),  # Longitude
        10.0,  # Altitude
        0  # Mission type
    ))

    # Fly to the second waypoint (sequence number 2 now)
    mission_items.append(mission_raw.MissionItem(
        2,  # Sequence number
        3,  # MAV_FRAME_GLOBAL_RELATIVE_ALT
        16,  # MAV_CMD_NAV_WAYPOINT
        0,  # Set current command
        1,  # Auto-continue
        0.0,  # Minimum pitch
        0.0,  # Empty param
        0.0,  # Empty param
        0,  # Yaw angle
        int(-35.3614586 * 1e7),  # Latitude
        int(149.1711402 * 1e7),  # Longitude
        25.0,  # Altitude
        0  # Mission type
    ))

    # RTl after mission is complete
    mission_items.append(mission_raw.MissionItem(
        3,  # Sequence number
        3,  # MAV_FRAME_GLOBAL_RELATIVE_ALT
        20,  # MAV_CMD_NAV_RETURN_TO_LAUNCH
        0,  # Set current command
        1,  # Auto-continue
        0.0,  # Minimum pitch
        0.0,  # Empty param
        0.0,  # Empty param
        0,  # Yaw angle
        0,  # Empty latitude
        0,  # Empty longitude
        0.0,  # Empty altitude
        0  # Mission type
    ))

    # Upload mission
    print("-- Uploading mission")
    await drone.mission_raw.upload_mission(mission_items)
    print("-- Done")

    
    print("-- Downloading mission")
    downloaded_items = await drone.mission_raw.download_mission()
    print("-- Done")

    print(f"Mission size: {len(downloaded_items)}")
    print(downloaded_items)

    # Arm drone
    print("-- Arming")
    await drone.action.arm()

    # Set take off altitude
    print("-- Setting takeoff altitude")
    await drone.action.set_takeoff_altitude(10)

    # Takeoff
    #print("-- Taking off")
    await drone.action.takeoff()

    await asyncio.sleep(10)

    # Start mission
    print("-- Starting mission")
    await drone.mission_raw.start_mission()

if __name__ == "__main__":
    # Run the asyncio loop
    asyncio.run(run())

```

julianoes, 2024-10-13:
Sorry, I forgot to ask and just assumed: this is ArduPilot, right?

I don't think you need a takeoff command in 0, you just need something in 0, which is where you can put the home position.