mavlink/MAVSDK-Python #733: Parameter Error with `set_maximum_speed` and `get_maximum_speed`

Ziaad03, 2024-10-19:
I encountered an issue when using the `set_maximum_speed` and `get_maximum_speed` functions in MAVSDK. Whenever I attempt to use these methods, I receive a parameter error, and the command fails. I also want to mention that I am using Arduopilot.

julianoes, 2024-10-21:
You're right. Looks like it's not supported for ArduPilot in the current state. It's using a PX4 only param.

https://github.com/mavlink/MAVSDK/blob/963f05a4b443e4751d98f8c7555395b1505d3245/src/mavsdk/plugins/action/action_impl.cpp#L725-L731

https://github.com/mavlink/MAVSDK/blob/963f05a4b443e4751d98f8c7555395b1505d3245/src/mavsdk/plugins/action/action_impl.h#L130

Ziaad03, 2024-10-21:
Ok, do you have any idea how the speed can be changed other than using these methods?

julianoes, 2024-10-21:
Check the ArduPilot docs and parameters.

Ziaad03, 2024-10-22:
There is a parameter called `WPNAV_SPEED` that controls the drone speed in auto mode only but it only change the speed to a value under 10 m/s . If I try to give it a value above 10 , the speed will not change and stay at 10 m/s. Also, I want to say that I really appreciate your help, and I want to thank you for your efforts.

julianoes, 2024-10-22:
Thank you for your kind words.

I have to say I'm somewhat tempted to remove this MAVSDK API because:
- it is misleading for ArduPilot, as you just found
- it only works for PX4 multirotor but not for fixedwing, so it can again be misleading.