mavlink/MAVSDK-Python #737: px4 sitl gazebo standard vtol sim error fail safe

MuhammedEnesGurkan, 2024-11-29:
While heading to my waypoint, it doesn't pass through all the points and doesn't print them. Sometimes, when I run it, it goes into failsafe mode before reaching the waypoint. Also, it can't switch to Fixed Wing mode; when I include it in a mission in MAVSDK, sometimes all the motors work together, and sometimes only one motor works. However, before reaching the desired location, it circles in the air.

-------------------------------------------------------------
[22:38:41.772] Critical: Mission failure: unable to reach heading within timeout
[22:38:41.868] Critical: Failsafe activated: switching to RTL in 5 seconds
- Mission cannot be completed
- No manual control input

[22:38:46.879] Critical: Failsafe activated: switching to RTL
- Mission cannot be completed
- No manual control input

julianoes, 2024-11-30:
Is this PX4 or ArduPilot? What version? How are you trying all this? Any logs? I don't have nearly enough information to attempt to help you I'm afraid.

MuhammedEnesGurkan, 2024-11-30:
I use px4 and gazebo for simulation. I use classic gazebo vtol. I am using latest version mavsdk. Sometimes in qgc it says no manual control please check health first.

julianoes, 2024-12-02:
Please update a log to logs.px4.io and link it here. And describe what you tried to do and what happened instead.

MuhammedEnesGurkan, 2024-12-11:
[14:26:30.434] Info: Armed by external command
[14:26:32.433] Info: Executing Mission
[14:26:32.433] Info: Climb to 25.0 m above home
[14:26:32.435] Info: Takeoff detected
[14:27:14.198] Critical: Mission failure: unable to reach heading within timeout
[14:27:14.199] Critical: Failsafe activated: switching to RTL in 5 seconds
- Mission cannot be completed
- No manual control input

[14:27:18.533] Critical: Failsafe activated: switching to RTL
- Mission cannot be completed
- No manual control input

[14:27:18.537] Info: RTL: start return at 548 m (60 m above destination)

MuhammedEnesGurkan, 2024-12-11:
> [14:26:30.434] Info: Armed by external command [14:26:32.433] Info: Executing Mission [14:26:32.433] Info: Climb to 25.0 m above home [14:26:32.435] Info: Takeoff detected [14:27:14.198] Critical: [...]

I am using missions but i cannot complete because of this errors

julianoes, 2024-12-12:
You don't have manual control input. You either need to provide manual control input, or try to disable the check using the param [NAV_RCL_ACT](https://docs.px4.io/v1.15/en/advanced_config/parameter_reference.html#NAV_RCL_ACT).

MuhammedEnesGurkan, 2024-12-12:
what is manual control input is it rc controller?

MuhammedEnesGurkan, 2024-12-12:
16:41:41.406] Info: RTL: start return at 549 m (60 m above destination)
[16:42:39.698] Info: RTL: land at destination
[16:42:50.244] Info: logging: opening log file 2024-12-12/13_42_50.ulg
[16:43:05.767] Info: Armed by external command
[16:43:07.770] Info: Executing Mission
[16:43:07.770] Info: Climb to 25.0 m above home
[16:43:08.773] Info: Takeoff detected
[16:44:28.758] Critical: Mission failure: unable to reach heading within timeout
[16:44:28.782] Critical: Failsafe activated: switching to RTL in 5 seconds
- Mission cannot be completed
- No manual control input

[16:44:33.794] Critical: Failsafe activated: switching to RTL
- Mission cannot be completed
- No manual control input

[16:44:33.794] Info: RTL: start return at 549 m (60 m above destination) i am use this command     await iha.param.set_param_int("NAV_RCL_ACT", 0)
 but it says this.

MuhammedEnesGurkan, 2024-12-12:
I figured out this problem with dividing waypoints. But i have a question how can i find the default timeout sec for missions.

julianoes, 2024-12-12:
No idea what you mean and it sounds like a question to ask on the PX4 Discord or discuss.px4.io, not a MAVSDK issue to me.

closed 2024-12-12 by julianoes