mavlink/MAVSDK-Python #739: Moving from Dronekit to MAVSDK

kripper, 2024-12-07:
I'm writing a Dronekit wrapper to make it easier to migrate Dronekit apps and use MAVSDK.

1) There are many Mavlink msgs that are not supported.
For example, how can we send a `MAV_FRAME_LOCAL_NED` msg with MAVDK? 

```
msg = self.vehicle.message_factory.set_position_target_local_ned_encode(
	0,

	0,          # SysId
	0,          # CompId

	mavutil.mavlink.MAV_FRAME_LOCAL_NED,          # frame
	0b111,          # mask (ignore x,y,z)

	0,	        # x
	0,	        # y
	0,          # z

	vx, 		# vx
	vy,		    # vy
	vz,         # vz

	0,          # afx
	0,          # afy
	0,          # afz

	0,          # yaw
	yawRate,          # yaw rate
)
self.vehicle.send_mavlink(msg)
self.vehicle.flush()
```

2) Is there any way to send a custom message built with `message_factory` through the MAVSDK connection?
This would be very handy (feature request).

julianoes, 2024-12-09:
MAVSDK in C++ allows for any message to be sent using the MavlinkPassthrough plugin. However, since that pulls in the MAVLink C headers, that's not currently supported in the language wrappers such as Python.

My plan is to add support for any messages in v4 (v3 is about to come out). It will allow to set and send custom messages at runtime, rather than compile time, along the lines of:

```
message = mavlink_passthrough.create_message("MESSAGE_NAME")
message.set_field_float("altitude", 1.0)
mavlink_passthrough.send_message(message)
...

Now for your request, have you looked into the offboard plugin? I think it has an API that is pretty close to it:
https://github.com/mavlink/MAVSDK/blob/bc0da1ea516bb0a015052aca1bf224cb2bd93326/src/mavsdk/plugins/offboard/offboard_impl.cpp#L381-L423

kripper, 2024-12-09:
Yes, I'm using offboard and I think I can live with some missing Mavlink messages for now.