mavlink/MAVSDK-Python #743: Takeoff altitude fixed to 2 [m]?

kripper, 2024-12-08:
We do:

```
await drone.action.set_takeoff_altitude(1)
await self.vehicle.system.action.takeoff()
```

But the drone is receiving a Mavlink message with param alt = 2 [m]

kripper, 2024-12-08:
This is the received mavlink msg:

[image]

julianoes, 2024-12-09:
Can you add that to https://github.com/mavlink/MAVSDK-Python/issues/743. Closing as duplicate.

closed 2024-12-09 by julianoes

kripper, 2024-12-09:
> Can you add that to #743

This here is 743 :-)
You probably mean another issue.

kripper, 2024-12-09:
Oops. Sorry. No idea how it got posted twice.

You meant https://github.com/mavlink/MAVSDK-Python/issues/742#issuecomment-2526729351