mavlink/MAVSDK-Python #750: Get GPS Global origin in telemte

alireza787b, 2025-02-12:
Hi,
 I'm trying to retrieve the GPS coordinates of the local NED origin that PX4 uses. I'd like to access the latitude, longitude, and altitude (AMSL) of this origin point.
According to the cpp documentation, the Telemetry object should have a gps_global_origin() method to achieve this. However, when I try to use it, I encounter an AttributeError: 'Telemetry' object has no attribute 'gps_global_origin'.
Here's a snippet of the code I'm using:
from mavsdk import System

async def get_gps_global_origin(drone: System):
    async for origin in drone.telemetry.gps_global_origin():
        print(origin)
        break
And here's the traceback I receive:
text
Traceback (most recent call last):
  File "/root/mavsdk_drone_show/drone_show.py", line 801, in pre_flight_checks
    async for position in drone.telemetry.gps_global_origin():
AttributeError: 'Telemetry' object has no attribute 'gps_global_origin'
I'm wondering if I'm misunderstanding how to access this information or if there might be an issue with the current MAVSDK implementation or this method is not implemnted in python. Could you please provide some guidance on the correct way to obtain the GPS Global Origin using MAVSDK Python?
Any help or insights would be greatly appreciated!

julianoes, 2025-02-12:
What about this one?
http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/plugins/telemetry.html#mavsdk.telemetry.Telemetry.get_gps_global_origin

alireza787b, 2025-02-16:
I saw this in docs but I got an error that it was not defined! maybe a mavsdk version problem. I will check that agian. thanks

julianoes, 2025-02-16:
You're trying to use `gps_global_origin()` but it's `get_gps_global_origin()`.

alireza787b, 2025-02-17:
here is where I tested that
                    home_position = drone.telemetry.get_gps_global_origin()

on line 802
https://github.com/alireza787b/mavsdk_drone_show/commit/2bb57c9e6d85a352aafc61c93c45eb462f9d3346#diff-d24a66ceef6c112f00ea91b75ab3bcb363f854ae63d636fe72114ef6c49274fdR802

but honestly, I didn't try hard with that. I did a walkaround and used mavlink2rest and pymavlink in another part of my code base and shared its value to fix this problem. That's why I think maybe sth was wrong with my syntax or the mavsdk version was onboard that drone.