mavlink/MAVSDK-Python #751: set_manual_control_input is not doing any action

husanboyqodirov, 2025-02-21:
Hi, I'm using SITL on Mission Planner.

My source code is working well for actions such as "Arm, Disarm, Take off, Land".

But controll actions are not working:

`async def send_control(pitch: float, roll: float, throttle: float, yaw: float, duration: float = 0.5):
    """
    Sends manual control input for a specified duration while preserving current throttle.
    """
    global current_throttle
    # Ensure throttle is not reset unexpectedly
    if throttle == 0:
        throttle = current_throttle  # Use the last known throttle value

    print(f"Sending control input - Pitch: {pitch}, Roll: {roll}, Throttle: {throttle}, Yaw: {yaw}")

    # Make sure manual control is enabled
    await drone.manual_control.start_position_control()

    # Send the control commands in intervals
    for _ in range(int(duration / 0.1)):  # Send command every 0.1s for the given duration
        await drone.manual_control.set_manual_control_input(pitch, roll, throttle, yaw)
        await asyncio.sleep(0.1)

    # Update current throttle for the next command
    current_throttle = throttle

async def throttle_up():
    """Increases throttle (altitude)."""
    global current_throttle
    current_throttle = min(1.0, current_throttle + 0.1)  # Increase throttle with a limit of 1.0
    await send_control(0, 0, current_throttle, 0, 1)  # Increase throttle for 1 second
........`

julianoes, 2025-03-11:
Check this: https://github.com/mavlink/MAVSDK/issues/2507#issuecomment-2654885076

husanboyqodirov, 2025-03-22:
Resolved. It turns out the same code is working well for PX4 firmware but not for Ardupilot.

closed 2025-03-22 by husanboyqodirov

husanboyqodirov, 2025-07-20:
@Surya-prathamzzz Which firmware are you using?

Surya-prathamzzz, 2025-07-21:
@husanboyqodirov currently  I am getting no error earlier the error was INPUT_NOT_SET then passing 0 value while starting manual control in the connect button then got INPUT_OUT_OF_RANGE when throttle value was -1 to 1 then changed it to 0 to 1 
when I am arming the drone using button the drone is getting armed but by changing throttle value there is no effect on the drone at all and it is getting disarmed 

I used these 2 libs
    implementation 'io.mavsdk:mavsdk:2.1.0'
    implementation 'io.mavsdk:mavsdk-server:2.1.6'
and this is current code for my android app where I am connecting drone and the arm disarm buttons are working 

package com.example.joystickmavsdk;

import android.os.Bundle;
import android.os.Handler;
import android.util.Log;
import android.view.InputDevice;
import android.view.MotionEvent;
import android.view.View;
import android.widget.Button;
import android.widget.TextView;
import android.widget.Toast;

import androidx.appcompat.app.AppCompatActivity;

import io.mavsdk.System;
import io.mavsdk.manual_control.ManualControl;
import io.mavsdk.action.Action;
import io.reactivex.android.schedulers.AndroidSchedulers;
import io.reactivex.schedulers.Schedulers;

public class MainActivity extends AppCompatActivity {

    private System drone;
    private ManualControl manualControl;
    private Action action;

    private float roll = 0f;
    private float pitch = 0f;
    private float yaw = 0f;
    private float throttle = 0f;

    private TextView joystickText;
    private final Handler handler = new Handler();

    private boolean isSendingControl = false;

    @Override
    protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);
        setContentView(R.layout.activity_main);

        joystickText = findViewById(R.id.tv_joystick);
        Button btnConnect = findViewById(R.id.btn_connect);
        Button btnArm = findViewById(R.id.btn_arm);
        Button btnDisarm = findViewById(R.id.btn_disarm);
        Button btnTakeoff = findViewById(R.id.btn_takeoff);

        btnConnect.setOnClickListener(v -> connectToDrone());
        btnArm.setOnClickListener(v -> armDrone());
        btnDisarm.setOnClickListener(v -> disarmDrone());
        btnTakeoff.setOnClickListener(v -> takeoffDrone());
    }
    private void connectToDrone() {
        MAVSDKServerManager serverManager = new MAVSDKServerManager();
        drone = serverManager.getDrone();
        action = drone.getAction();
        manualControl = drone.getManualControl();

        // Step 1: Send initial input before starting control mode
        manualControl.setManualControlInput(0.0f, 0.0f, 0.0f, 0.0f)
                .subscribeOn(Schedulers.io())
                .observeOn(AndroidSchedulers.mainThread())
[... 73 lines cut ...]
        if (range == null) return 0;

        float value = event.getAxisValue(axis);
        float flat = range.getFlat();
        return Math.abs(value) > flat ? value : 0;
    }

    private void startSendingManualControl() {
        if (isSendingControl) return;
        isSendingControl = true;

        handler.postDelayed(new Runnable() {
            @Override
            public void run() {
                if (manualControl != null) {
                    manualControl.setManualControlInput(roll, pitch, throttle, yaw)
                            .subscribe(() -> {}, throwable -> Log.e("ManualInput", "Failed", throwable));

                    Log.d("SEND_INPUT", String.format("r: %.2f p: %.2f t: %.2f y: %.2f", roll, pitch, throttle, yaw));
                }
                handler.postDelayed(this, 40); // ~25 Hz
            }
        }, 40);
    }
}

Surya-prathamzzz, 2025-07-22:
> [@Surya-prathamzzz](https://github.com/Surya-prathamzzz) Which firmware are you using?

my drone has ardupilot flight controller @husanboyqodirov