mavlink/MAVSDK-Python #755: Incorrect Health-information with MAVSDK-Python 3.0.1

ans1kel, 2025-04-02:
Using the latest version (MAVSDK-Python 3.0.1), I'm getting some incorrect values in `mavsdk.telemetry.Health` from a PX4 device

### Erroneous behavior
For example, the `is_accelerometer_calibration_ok` field value is False in the following:
`Health: [is_gyrometer_calibration_ok: True, is_accelerometer_calibration_ok: False, is_magnetometer_calibration_ok: True, is_local_position_ok: True, is_global_position_ok: False, is_home_position_ok: False, is_armable: False]`

Sometimes, also the `is_gyrometer_calibration_ok` and/or `is_magnetometer_calibration_ok` are showing incorrect values.

### Expected behavior
When running an earlier version (MAVSDK-Python 1.4.9) the field value is correct:
`Health: [is_gyrometer_calibration_ok: True, is_accelerometer_calibration_ok: True, is_magnetometer_calibration_ok: True, is_local_position_ok: True, is_global_position_ok: False, is_home_position_ok: False, is_armable: False]`

### Environment
- MAVSDK-Python
  - Python 3.10.12
  - running in Ubuntu 22.04 -based container
  - using Docker version 24.0.5 (x86_64)
- PX4
  - version 1.15.4
  - Holybro Durandal v1

[mavsdk-python-1.4.9.log](https://github.com/user-attachments/files/19567009/mavsdk-python-1.4.9.log)
[mavsdk-python-3.0.1.log](https://github.com/user-attachments/files/19567008/mavsdk-python-3.0.1.log)

julianoes, 2025-04-04:
This is probably related to https://github.com/mavlink/MAVSDK/pull/2521 and we should update the Python wrapper...

julianoes, 2025-04-04:
I'm doing a release but I have to work through everything that broke in CI...

https://github.com/mavlink/MAVSDK-Python/pull/756

alexandre-bonnefond, 2025-12-02:
Is there any update on this bug? I am still receiving incorrect flags with "mavsdk==3.10.2"

alexandre-bonnefond, 2025-12-05:
@julianoes thanks for the support! I actually found out where my error is coming from. In my script I fetch the health message once and read its outputs. The problem is that the calibration flags value are set to False at first and only update after some time. I saw that by using your method with the `telemetry.py` and adding the health in it, you can see the logs below. Is this behavior normal?

Overall this solves my problem, I will just add a delay in my script. 

Thank you.

```
INFO:mavsdk_server:MAVSDK version: v3.10.2 (mavsdk_impl.cpp:32)                                                                                                                                
INFO:mavsdk_server:Waiting to discover system on serial:///dev/pixhawk:2000000... (connection_initiator.h:20)                                                                                  
WARNING:mavsdk_server:Vehicle type changed (new type: 2, old type: 0) (system_impl.cpp:323)                                                                                                    
INFO:mavsdk_server:System discovered (connection_initiator.h:62)                                                                                                                               
INFO:mavsdk_server:Server started (grpc_server.cpp:177)                                                                                                                                        
INFO:mavsdk_server:Server set to listen on 0.0.0.0:50051 (grpc_server.cpp:178)                                                                                                                 
Health: [is_gyrometer_calibration_ok: False, is_accelerometer_calibration_ok: False, is_magnetometer_calibration_ok: False, is_local_position_ok: True, is_global_position_ok: False, is_home_p
osition_ok: False, is_armable: False]                                                                                                                                                          
Health: [is_gyrometer_calibration_ok: False, is_accelerometer_calibration_ok: False, is_magnetometer_calibration_ok: False, is_local_position_ok: True, is_global_position_ok: False, is_home_p
osition_ok: False, is_armable: False]                                                                                                                                                          
Health: [is_gyrometer_calibration_ok: False, is_accelerometer_calibration_ok: False, is_magnetometer_calibration_ok: False, is_local_position_ok: True, is_global_position_ok: False, is_home_p
osition_ok: False, is_armable: False]                                                                                                                                                          
Health: [is_gyrometer_calibration_ok: False, is_accelerometer_calibration_ok: False, is_magnetometer_calibration_ok: False, is_local_position_ok: True, is_global_position_ok: False, is_home_p
osition_ok: False, is_armable: False]                                                                                                                                                          
Health: [is_gyrometer_calibration_ok: False, is_accelerometer_calibration_ok: False, is_magnetometer_calibration_ok: False, is_local_position_ok: True, is_global_position_ok: False, is_home_p
osition_ok: False, is_armable: False]                                                                                                                                                          
Health: [is_gyrometer_calibration_ok: False, is_accelerometer_calibration_ok: False, is_magnetometer_calibration_ok: False, is_local_position_ok: True, is_global_position_ok: False, is_home_p
osition_ok: False, is_armable: False]                                                                                                                                                          
Health: [is_gyrometer_calibration_ok: False, is_accelerometer_calibration_ok: False, is_magnetometer_calibration_ok: False, is_local_position_ok: True, is_global_position_ok: False, is_home_p
osition_ok: False, is_armable: False]                                                                                                                                                          
Health: [is_gyrometer_calibration_ok: False, is_accelerometer_calibration_ok: False, is_magnetometer_calibration_ok: False, is_local_position_ok: True, is_global_position_ok: False, is_home_p
osition_ok: False, is_armable: False]                                                                                                                                                          
Health: [is_gyrometer_calibration_ok: False, is_accelerometer_calibration_ok: False, is_magnetometer_calibration_ok: False, is_local_position_ok: True, is_global_position_ok: False, is_home_p
osition_ok: False, is_armable: False]                                                                                                                                                          
Health: [is_gyrometer_calibration_ok: False, is_accelerometer_calibration_ok: False, is_magnetometer_calibration_ok: False, is_local_position_ok: True, is_global_position_ok: False, is_home_p
osition_ok: False, is_armable: False]                                                                                                                                                          
[... cut ...]
Health: [is_gyrometer_calibration_ok: True, is_accelerometer_calibration_ok: True, is_magnetometer_calibration_ok: False, is_local_position_ok: False, is_global_position_ok: False, is_home_po
sition_ok: False, is_armable: False]                                                                                                                                                           
Health: [is_gyrometer_calibration_ok: True, is_accelerometer_calibration_ok: True, is_magnetometer_calibration_ok: False, is_local_position_ok: True, is_global_position_ok: False, is_home_pos
ition_ok: False, is_armable: False]
```

julianoes, 2025-12-06:
That makes sense. I guess you have to wait a bit, or just subscribe to it until it is ready. It makes sense if the vehicle is still initializing everything.

closed 2025-12-06 by julianoes