mavlink/MAVSDK-Python #757: timeout on arm and takeoff using mavserver

end-of-small-sanctuary, 2025-04-07:
hello again, sorry for necrobumping another issue (new to github etiquette).

as per my comment [here ](https://github.com/mavlink/MAVSDK-Python/issues/217) and @julianoes 's request, here's my mavserver log.

[mavserver log.txt](https://github.com/user-attachments/files/19630944/mavserver.log.txt)

i waited until all initial messages stopped, then ran my code again while the drone was "ready to fly" in qgroundcontrol.
the voltage error shouldn't be the issue, as this happens even when the error doesn't appear.

this issue still occurs when i manually arm the drone with qgroundcontrol and try takingoff with mavserver.

sorry for the late reply, laptop overheated when I tried to get the log 2 days ago.

julianoes, 2025-04-08:
It doesn't look like any command actually arrives, right? They all time out.

What is the connection like?

And also, this doesn't look right:

```
udpin://127.0.0.1:50051
```

You are using what is usually a gRPC port and use it as the MAVLink communication port.

Please share the code where you set up the connection.

end-of-small-sanctuary, 2025-04-08:
after setting up qgroundcontrol to forward to port 50051, i run this in powershell:

.\mavsdk_server_bin.exe -udpin udpin://127.0.0.1:50051

is that it? I've established a connection with the drone in python, it's in the code i commented on in the previous issue.

julianoes, 2025-04-08:
Don't use port 50051 like that. That's the port used by gRPC between MAVSDK-Python and mavsdk_server, at least by default.

end-of-small-sanctuary, 2025-04-08:
is there any recommended ports? 

would 14550 be appropriate?

end-of-small-sanctuary, 2025-04-08:
i tried `.\mavsdk_server_bin.exe -udpin udpin://127.0.0.1:14580 ` after changing the forwarding port, which successfully connected mavserver and qgroundcontrol, but it refused to connect to the server when i changed the port in the code from 50051 to 14580.

when i continue as usual, with the server running on 14580 and the code port unchanged from 50051, the code runs normal but it yields the same result of arm timing out.

its currently like this: 
  ` drone = System(mavsdk_server_address='127.0.0.1', port=50051)` 
    ` print("MAVSDK System initialized.") ` 

   `  await drone.connect(system_address="udp://:14540") `

julianoes, 2025-04-08:
You have to tell me how you actually connect. I don't know where your mavlink is coming from.

end-of-small-sanctuary, 2025-04-08:
```
You have to tell me how you actually connect. I don't know where your mavlink is coming from.
```

I'm not sure i understand what you mean, im connecting to a IRL drone via Sik radio telemetry.

here's my step-by-step process:

1-turn on drone.
2-run mavserver on 14580 port.
3-run qgroundcontrol.
4-run python code when the drone is ready to fly.

the MavLink would be coming from qgroundcontrol no?

julianoes, 2025-04-08:
> drone via Sik radio telemetry.

That's what I needed to know.

Right, so if you want a connection to QGC and MAVSDK, you have a few ways to do that:

- You can either enable forwarding in QGC: https://docs.qgroundcontrol.com/master/en/qgc-user-guide/settings_view/mavlink.html#ground_station The docs say this is one-way, if that's correct, then MAVSDK would not be able to talk to PX4, so that's no good.
- Create a MAVSDK forwarder that splits the traffic from serial/USB to two UDP ports. You can see example code here: https://github.com/mavlink/MAVSDK/tree/main/examples/mavlink_forwarding, and some docs: https://mavsdk.mavlink.io/main/en/cpp/guide/connections.html#forwarding-mavlink-between-connections
- You can use a tool like mavlink-router (https://github.com/mavlink-router/mavlink-router) to do the same.

end-of-small-sanctuary, 2025-04-13:
so i've tried to do as you've said (hopefully correctly)
1-**launch mavlin_forwarded.py**: which is this
 `import socket
import serial
from pymavlink import mavutil

# === Configuration ===
SERIAL_PORT = 'COM6'         # Change to your telemetry serial port
BAUDRATE = 57600             # Typical for SiK radios
UDP_TARGETS = [
    ('127.0.0.1', 14540),    # QGroundControl
    ('127.0.0.1', 14541),    # MAVSDK Python app
]

def main():
    # Open serial connection
    ser = serial.Serial(SERIAL_PORT, BAUDRATE, timeout=0)

    # Setup UDP sockets
    udp_sockets = []
    for ip, port in UDP_TARGETS:
        sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
        udp_sockets.append((sock, (ip, port)))

    print(f"Forwarding MAVLink from {SERIAL_PORT} to {[f'{ip}:{port}' for ip, port in UDP_TARGETS]}")

    while True:
        if ser.in_waiting:
            data = ser.read(ser.in_waiting)
            for sock, addr in udp_sockets:
                sock.sendto(data, addr)

if __name__ == '__main__':
    try:
        main()
    except KeyboardInterrupt:
        print("\nStopped by user.")`

2- **launch mavserver -p 50051**
3-**when launching qgc, connect to forwared udp in comm link settings**: i know this works when i forward packets since it shows the location of the drone (no status)
4-**launch pytry (main arming code)**

launching the arming code gives me the timeout error, and it looks like the same error in mavserver.

im seriously stumped.

[mavserver log2.txt](https://github.com/user-attachments/files/19723279/mavserver.log2.txt)

julianoes, 2025-04-13:
I guess you went with your own forwarding solution. So you expect me to test what you came up with? 😄

end-of-small-sanctuary, 2025-04-13:
you're right, that was wrong of me.
the solutions you provided are all C++, is there anyway i can use python instead?

end-of-small-sanctuary, 2025-04-13:
so i solved the issue:
 running `.\mavsdk_server_bin.exe --forwarding serial://COM6:57600 -p 50051`

and using 
```
drone = System(mavsdk_server_address='localhost', port=50051)
    print("MAVSDK System initialized.")

    # Connect to the drone via MAVLink UDP port (QGroundControl/telemetry should send here)
    await drone.connect() 

```
 for the mavsdk connection

unfortunately this means that i can't use qgc, do with that what you will.

closed 2025-04-13 by end-of-small-sanctuary

julianoes, 2025-04-13:
Did you add the argument `--forwarding` to mavsdk_server, or did an LLM invent it?

Because I don't see it: https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk_server/src/mavsdk_server_bin.cpp

The solutions to "forward/route" MAVLink that I gave you are not C++ as such, they are software that you can install and use, e.g. [mavlink-router](https://github.com/mavlink-router/mavlink-router), or if you want it in Go, use [mavp2p](https://github.com/bluenviron/mavp2p), so really these are standalone tools next to your Python scripting.

end-of-small-sanctuary, 2025-04-14:
When i read the ForwardingOption::ForwardingOn flag in the documentation, I thought that since mavserve is already a forwarder in a sense (gets data from qgc, forwards it through port 50051) it might also do the same with serial.

so i asked an LLM for an equivalent flag to (ForwardingOption::ForwardingOn) that i can use for mavserver cli, and it gave me --forwarding, and it worked. 

i _just_ did another run with this command: `.\mavsdk_server_bin.exe  serial://COM6:57600 ` and it also worked, so --forwarding wasn't doing anything anyway in my case.

im trying to minimise the number of processes run, so using more tools would not be ideal, thats why i asked for solutions in Python (i know, beggers cant be choosers) to see if i can run them alongside the main script in a senes.

Thank you so much for your time, julian! you're a saint!

julianoes, 2025-04-14:
I like small tools that do one thing, rather than have one tool want to do it all. Hence no forwarding included within mavsdk_server.