mavlink/MAVSDK-Python #760: mavsdk_server_bin.exe does not work on windows 11

Cheongmaesil, 2025-05-09:
I tried to send real-time Qualisys data to a Pixhawk as mocap input from a Windows desktop using telemetry.
However, when I run mavsdk_server_bin.exe, it just shuts down immediately.
How can I fix this issue?
Please help me.

julianoes, 2025-05-11:
What is the output? What MAVSDK version? Without more information it's hard to help you.

Cheongmaesil, 2025-05-12:
I am using Python 3.9.21, pymavlink 2.4.43, and MAVSDK 3.0.1 in an Anaconda environment on a 64-bit Windows system. I have also installed and checked out the MAVSDK binary server (versions 3.1.0 through 3.3.3). Please let me know if you need any further information—I will get back to you promptly.

julianoes, 2025-05-12:
If you start the binary manually, so not via Python, what is the output?

Cheongmaesil, 2025-05-14:
[image]

[image]

When I ran my code about qtm_mavsdk, I saw the following message in the debug terminal (first image).
So, of course, I manually ran the binary as admin. However, in second the image, there's no output at all...

julianoes, 2025-05-14:
Odd. What's the return value?

I'll double check it works on my Windows 11.

Cheongmaesil, 2025-05-15:
[image]

I got the error message. Maybe it is helpful for you...

julianoes, 2025-05-15:
I have no idea, I have to ask claude.ai:

> What does it mean if a windows program exits without any output and return code -1073741819EditWhen a Windows program exits with return code -1073741819 (which is equivalent to 0xC0000005 in hexadecimal), it indicates a memory access violation error. This is a common and significant error in Windows applications.
This error means the program attempted to read from or write to a memory location that it doesn't have permission to access, or tried to access memory that doesn't exist. It's essentially the Windows equivalent of a segmentation fault in Unix/Linux systems.
Common causes include:

> Dereferencing null pointers
Buffer overflows
Stack overflows
Accessing memory that has been freed
Array index out of bounds
Corrupted heap memory

> The fact that the program exited without any output suggests the crash happened before it could display any diagnostic information. This often happens with critical memory errors that cause immediate termination.
To troubleshoot this issue, you could:

> Check application logs in Event Viewer
Run the program with a debugger attached
Look for Windows Error Reporting (WER) files
Try running the program with administrator privileges (if appropriate)
Check for updates to the program or Windows itself
Scan for malware that might be corrupting memory

> This type of error typically requires debugging at the code level to properly diagnose and fix.

Cheongmaesil, 2025-05-17:
I installed MAVSDK via pip and then cloned the MAVSDK-Python repository.
In my QTM–MAVSDK Python script, I simply use:

    import mavsdk

Are there any additional steps I need to take after cloning?  
For example, do I need to build MAVSDK with CMake or run:

    pip install -r requirements.txt

Cheongmaesil, 2025-05-22:
Thank you, I'm going to try it as soon as possible!!