mavlink/MAVSDK-Python #763: Protobuf Warning (UserWarning: Protobuf gencode version 5.29.0)

afwilkin, 2025-06-03:
[Video Describing Warning](https://www.loom.com/share/5ff456b20a3b48d5968643b0fc32ae5c)

It SEEMS like protobuf python version was just upgraded and the .proto files seem to be mismatched by a major version

`/home/atlas/iperf_testing/test_dep/env/lib/python3.10/site-packages/google/protobuf/runtime_version.py:98: UserWarning: Protobuf gencode version 5.29.0 is exactly one major version older than the runtime version 6.31.1 at mission/mission.proto. Please update the gencode to avoid compatibility violations in the next runtime release.
  warnings.warn(
/home/atlas/iperf_testing/test_dep/env/lib/python3.10/site-packages/google/protobuf/runtime_version.py:98: UserWarning: Protobuf gencode version 5.29.0 is exactly one major version older than the runtime version 6.31.1 at mission_raw/mission_raw.proto. Please update the gencode to avoid compatibility violations in the next runtime release.
  warnings.warn(
/home/atlas/iperf_testing/test_dep/env/lib/python3.10/site-packages/google/protobuf/runtime_version.py:98: UserWarning: Protobuf gencode version 5.29.0 is exactly one major version older than the runtime version 6.31.1 at mission_raw_server/mission_raw_server.proto. Please update the gencode to avoid compatibility violations in the next runtime release.
`

To recreate:

- pip3 install mavsdk
- -try to run example here: https://github.com/mavlink/MAVSDK-Python/tree/main/examples

afwilkin, 2025-06-03:
This could also be a me thing that I am not seeing, but thought i would at least mention it

afwilkin, 2025-06-11:
@julianoes I havent manually updated it. but, when i start a new project and pip3 install mavsdk
it seems to install the latest protobuf

[image]

I would say that if you are in a new environment with a virtual environment...probably a me problem. 

I am on ubuntu 22.

julianoes, 2025-06-12:
@jack-mil feel free to make a pull request to fix it. You seem to have a better handle on it than me. I haven't had time to look into it yet.

jack-mil, 2025-06-16:
Sorry, I am not familiar enough with this package, protobuf, or your build system in general to be able to make changes like that. I am not a regular user of mavsdk. But some sort of dependency lock file (uv.lock, frozen pip requirements, pylock.toml) for your Actions build workflow would help with pulling in unexpected dependency versions as build time.

closed 2025-09-17 by julianoes