mavlink/MAVSDK-Python #764: Can’t make MAVSDK commuincate with PX4 SITL in Airsim (Unreal Engine)

NikitaBlck, 2025-06-11:
I am running PX4 SITL in WSL on Windows and using it with AirSim to simulate a drone in Unreal Engine. When I am running a script with MAVSDK library, it says that I need to install MAVSDK server separately, but this server, when running, can’t connect to PX4 SITL, and there is no clear guide on what kind of IP address and port is needed for the MAVSDK server (WSL IP or something else, TCP or UDP). Please let me know if anyone has had this problem or a similar situation.

[image]
[image]
[image]
[image]
[image]

ne246, 2025-06-30:
@julianoes this is actually the exact issue im running into, can I just add on to this?

julianoes, 2025-06-30:
Read what I wrote in https://github.com/mavlink/MAVSDK-Python/issues/764#issuecomment-2964734348.

ne246, 2025-07-11:
@NikitaBlck DId you fix this issue? I cant seem to fix it.

closed 2025-09-03 by julianoes