mavlink/MAVSDK-Python #800: Connecting 2 sitl gazebo (docker) to qGC shows only one vehicle

pushkarprasad007, 2025-09-13:
I created two docker containers like so for SITL:

```
docker run --rm -it \
  jonasvautherin/px4-gazebo-headless:1.16.0
```
and
```
docker run --rm -it \
  -p 14560:14550 \
  -p 14541:14540 \
  jonasvautherin/px4-gazebo-headless:1.16.0
```
For the latter, I made sure the param MAV_SIS_ID is set to 2. 

So my idea is to be able to see both on qGC, by adding comm link for 14560. However, I don't see a dropdown on the qGC.

I intend to use mavsdk to fly both so they appear in the qGC together. 

Kindly let me know if there's any obvious mistake I'm making.

JonasVautherin, 2025-09-13:
I think this is more of a px4-gazebo-headless question, I would redirect you there: https://github.com/JonasVautherin/px4-gazebo-headless/issues/39.

The quick answer is that it is not supported, but I would be open to contributions. In your case, maybe you would want to look into simulating two drones in the same px4-gazebo-headless instance (maybe using mavlink-router to separate the streams outside of the container if necessary).

What you are trying to do above won't work, though.

I'll close it here, feel free to continue the discussion in https://github.com/JonasVautherin/px4-gazebo-headless/issues/39!

closed 2025-09-13 by JonasVautherin

pushkarprasad007, 2025-09-14:
> In your case, maybe you would want to look into simulating two drones in the same px4-gazebo-headless instance (maybe using mavlink-router to separate the streams outside of the container if necessary).

How would the ports be exposed outside of the container?